iNavFlight/inav-configurator #2231: Idea: Virtual Pitot enabled by default for Fixed Wing

chasdabigone, 2024-11-07:
I didn't know virtual pitot existed until I read about it explicitly. A few of my friends also did not know about it. Is there any reason it shouldn't be enabled by default? I think it would be a good change

MrD-RC, 2024-11-07:
One other option may be to enable virtual pitot if the pilot sets up a GPN in the setup wizard. What do you think @DzikuVx ?

chasdabigone, 2024-11-07:
maybe when you select "GPS for navigation and telemetry" it can turn virtual pitot on if it is NONE

MrD-RC, 2024-11-10:
> maybe when you select "GPS for navigation and telemetry" it can turn virtual pitot on if it is NONE

This would not be a good idea. Because someone could have set up a real airspeed sensor before enabling GPS. In the wizard, it would be before anything else happens.

chasdabigone, 2024-11-10:
That is why I thought it should check to make sure it is none

---------- Forwarded message ---------

default for Fixed Wing (Issue #2231)
To: iNavFlight/inav-configurator ***@***.***>
Cc: chas s ***@***.***>, Author ***@***.***>

closed 2024-11-15 by b14ckyy