iNavFlight/inav-configurator #2236: Demo mode gives 'callback is not a function' error when connecting

chasdabigone, 2024-11-08:
```
Running: - OS: Linux, Chrome: 120.0.6099.216, Configurator: 8.0.0
```

When I select "Demo mode" from the drop-down menu and click "Connect" the log says
```
Unexpected error: callback is not a function
Disconnecting...
```

MrD-RC, 2024-11-08:
Demo mode does not work in master. It will only work in releases. This is known.

stronnag, 2024-11-08:
> Demo mode does not work in master. It will only work in releases. This is known.

Out of interest why?
I have master installed; I have a SITL executable. It fails before it gets to run it.

stronnag, 2024-11-08:
In fact if you just comment out the (*seems to JS neophyte*) "bogus" callback line, then it just works.
Possibly yet  another artefact of the Electron migration.

chasdabigone, 2024-11-08:
> In fact if you just comment out the (*seems to this JS neophyte*) "bogus"

What is the best way to test changes i made to configurator without
plugging in a FC in master?

stronnag, 2024-11-09:
> What is the best way to test changes i made to configurator without plugging in a FC in master?

SITL

closed 2024-11-09 by chasdabigone

2A42, 2025-05-05:
> In fact if you just comment out the (_seems to this JS neophyte_) "bogus" callback line, then it just works. Possibly yet another artefact of the Electron migration.

hey, please, could you point the location of this callback line?