iNavFlight/inav-configurator #2251: 8.0-RC1 Storing Geozones causes all free number fields to be read-only until restart

b14ckyy, 2024-11-17:
## Reproduce

- Connect Configurator to FC
- Open Mission Control
- Create any Geozone if none present yet
- Click on Store to FC and wait for reboot
- Wait for reconnect
Result: No text field can be directly edited like Altitudes, Distances etc. 

Reconnect of FC does not fix
As a workaround, Configurator needs to be restarted entirely. 
Normal Save and Reboot from other pages are not trigger the issue as far as I can tell.

stronnag, 2024-11-17:
Does not happen here. After reboot, text fields are editable ... at least with the SITL as FC. Let me try a real one.

b14ckyy, 2024-11-17:
I can replicate on 2 FCs and also got the same report from 2 users. 
Are you on Windows or linux? Maybe only windows affected?

stronnag, 2024-11-17:
Linux. 
However, this looks strange ...
[image]

stronnag, 2024-11-17:
There is no problem on Linux with editing fields after reboot with "real" FCs, so maybe it's a Windows issue.

b14ckyy, 2024-11-17:
Yeah I think there is a confusion between ASL reference and Launch Alt reference for the validity check. 
[image]

@Scavanger

Scavanger, 2024-11-19:
Fix for sanity check: https://github.com/iNavFlight/inav-configurator/pull/2252

The bug that the fields (affects ALL(!) input fields even in other tabs!) are inactive after saving is strange, and as @stronnag said, only occurs under windows. I can't find a bug in the code. I haven't changed anything in the code and I guess it worked before. 
Apparently the input fields lose their focus. 
Workaround: Deactivate the Configurator window briefly (minimise, activate another window, etc.) then everything works again.

b14ckyy, 2024-11-19:
Fixed in https://github.com/iNavFlight/inav-configurator/pull/2252

closed 2024-11-19 by b14ckyy