iNavFlight/inav-configurator #2260: 8.0.0-RC1 -problematic Save to File behavior

pitts-mo, 2024-11-20:
inav-configurator 8.0.0-RC1 is persistently opening Save to File in the users $HOME path. And, after the first Save to File attempt each new attempt to Save to File is silently opening the save browser in the background.

This is observed in ubuntu 24.04 running from nightly builds last week and the 8.0.0-RC1 asset INAV-Configurator_linux_x64_8.0.0.deb. 

Reverting to INAV-Configurator_linux64_7.1.2.deb returns the behaviors of opening Save to File in the most recently utilized path and opening each attempt in a foreground window.

MrD-RC, 2024-11-21:
Already fixed in a patch for RC2.

MrD-RC, 2024-11-21:
https://github.com/iNavFlight/inav-configurator/pull/2258

mmosca, 2024-11-21:
@pitts-mo if you want to verify it, you can checkout the latest nightly build.

MrD-RC, 2024-11-21:
Not quite yet. It hasn't been merged. I think there are a couple of little issues to fix before merge. But it should be soon.

pitts-mo, 2024-12-15:
hmm,
I am still observing these same behaviors with https://github.com/iNavFlight/inav-configurator/tree/8.0.0-RC3

re-open this issue until resolved?

pitts-mo, 2025-01-24:
The behavior persists on final 8.0.0 release.

@MrD-RC Can you please re-open this issue.

mmosca, 2025-01-24:
@MrD-RC if you look into this, target maintenance-8.x.x for the pr, instead of master