iNavFlight/inav-configurator #2262: 8.0-20241120-5019b12 Missing Motor Numbers in Outputs tab for MR

b14ckyy, 2024-11-23:
The Outputs Tab misses the numbering on the Motor Graphic to identify motor positions and test order: 

[image]
[image]

mmosca, 2024-11-23:
Did it have that before? I think it never did. It can be helpful though.

mmosca, 2024-11-23:
> I thought it did at some point. Without them, identifying the motors can be hard as I need to screenshot or not down from the mixer page.

The motor numbers are usually marked on the esc and the mixer is needed to understand where they are in the quad, in case someone had it mixed up by rotating the esc.

I do think it is a good idea overall, but it needs to match what the mixer thinks it is, and potentially, have the motor order wizard here as well

b14ckyy, 2024-11-23:
Motor wizard is a neat feature but should get an update. Its a long time since I did with BF but I think there it twitches a motor with short pulses of 10% throttle and you click on a graphic what motor is spinning. This way the mixer is set up.

sensei-hacker, 2024-11-23:
Fixed in https://github.com/iNavFlight/inav-configurator/pull/2263

> I do think it is a good idea overall, but it needs to match what the mixer thinks it is, 

It does now. It used to have a static image with the default positions from the preset.
Then it had none. Now it has them correct, matching the mixer.

Caused by https://github.com/iNavFlight/inav-configurator/pull/1910
The preview image from the mixer was reused on the outputs tab.

closed 2024-11-27 by sensei-hacker