iNavFlight/inav-configurator #2264: Can not run configurator 8 on win7

and-sh, 2024-11-26:


and-sh, 2024-11-26:
Have old laptop on Windows 7. Instead of start configurator 8 sayd  DiscardVirtualMemory not found in kernel32.dll. 
Is configurator really need something in win10-11? 
It is unwise to take new laptop in field

mmosca, 2024-11-26:
Unfortunately, there is not much we can do here. Configurator 8 was updated from very old versions of the Chromium and moved to Electron framework and newer versions of Chromium require windows 10.

Maybe you can look into running some of the software that requires win7 on a virtual machine.

closed 2024-11-26 by mmosca

and-sh, 2024-11-26:
Eh oh
Just installed Debian as second boot. It seems to work.

End of life does not mean end of work. We have scientific instrument on Win98 with ISA bus control board. It works in realtime better than similar on win10.

sensei-hacker, 2024-11-26:
> Just to add extra information. Windows 7 reached end of life 10/01/2023. So almost 2 years.

Heck, mainstream support ended after 2014, ending January 2015. So almost TEN years ago, for Windows 7 itself.

Enterprises could buy extended support for a few years after that. SP was a few after that.

https://learn.microsoft.com/en-us/lifecycle/products/windows-7

sensei-hacker, 2024-11-26:
> Eh oh Just installed Debian as second boot. It seems to work.
> 
> End of life does not mean end of work. We have scientific instrument on Win98 with ISA bus control board. It works in realtime better than similar on win10.

Glad you got it working!
The local security guy is relieved you no longer have Windows 7, potentially connecting to the *internet*. 😁

wrobby493-cell, 2025-09-19:
This is very vital information to Traders who have been scammed before  my advice is for you to be wise before you invest in any binary options trade or broker  I was scammed for $122 000 USD by an online broker but at last I found someone who helped me to recover all my lost funds back from a scam that shocked my capital with an unregulated broker If you need assistance with regards of your lost funds from your broker kindly get in contact with  Recovery Expert He was the one who helped me to bring back my lost funds contact him Via email recoveryexpert326 at gmail dot com He will guide you on the steps I took in getting all my refunds and bonuses back Good Luck to you