iNavFlight/inav-configurator #2265: Total mess in configirator on small screens

and-sh, 2024-11-26:
1024x768 resolution cannot reach bottom save button, can not reach CLI.
Large font (accessibility) lead to worst results.
[image]

and-sh, 2024-11-26:
I found some solution. If I press Tab several times it switch  focus to buttons and they become partially visible. same with cli.

btw Linux is becoming noticeably more convenient than Windows. For example, I use different desktops for different tasks, which allows me to switch between them quite quickly.
[image]

MrD-RC, 2024-11-26:
I believe this has been fixed in 8.0.

and-sh, 2024-11-27:
This is 8

I guess that some minor parameters of window are incorrect. All content are in places. I can switch between them by Tab. I can not scroll by sliders only.

Found a new solution. Just stretch the window a little.
[image]
[image]

MrD-RC, 2024-11-27:
With a resolution that size, you should run Configurator maximised.

MrD-RC, 2024-11-28:
@and-sh Could you try using this configurator https://github.com/iNavFlight/inav-configurator/actions/runs/12065115088?pr=2266

and-sh, 2024-11-29:
On Debian, Gnome work fine, thank you very much
[image]

closed 2024-11-30 by MrD-RC