iNavFlight/inav-configurator #2268: After intalling MTF-01 optic flow althold does not work.

and-sh, 2024-11-29:
Inav 8.0.0
Althold was tuned before installing optflow. Worked normal.
After installing optflow.
Under approx 2m - work normal.
Higher does not work.
Testing in field over flat surface (snow). Sunny day.
Surface mode = OFF
nav_max_terrain_follow_alt = 100
inav_max_surface_altitude = 100

Max altitude where rangefinder measure reliable data 120cm.
[image]

Max altitude where rangefinder measure something about 3m.
[image]

About 16m rangefinder report zero altitude
[image]
At the same time surface reliability = 1000.
 
Log
[blackbox_log_2024-11-28_131957.TXT](https://github.com/user-attachments/files/17958253/blackbox_log_2024-11-28_131957.TXT)

and-sh, 2024-11-29:
I think the simplest way is do not report new altitude if distance strength or precision is low. Then agl engine decrease the surface reliability automatically. The same way it decrease surface reliability than altitude is higher than nav_max_terrain_follow_alt

sensei-hacker, 2024-11-29:
IMHO it shouldn't be using AGL at all if the pilot has AGL turned off (surface mode off).

If I choose to fly MSL altitude, it shouldn't kinda mix in a little AGL behind my back.

and-sh, 2024-11-29:
Rangefinder may be useful during landing because no matter how I isolate barometer there is high pressure close to the ground.
[image]

and-sh, 2024-11-29:
But I can land in surface mode instead.