iNavFlight/inav-configurator #2270: Multiple instances in Linux

cptX, 2024-12-01:
Hi, I'm using INAV in Linux (LM 22) and I would like to have multiple instances open in order to copy settings from one model to another.

**Current Behavior**

At the moment when I try to open another instance it just fails. If I run the binary from the terminal it says:
Opening in existing browser session.
libva error: vaGetDriverNames() failed with unknown libva error

**Desired Behavior**

Multiple instances/windows should be able to run. There is no reason not to as every instance can select its serial port independently, so I believe there will not be any conflict. If there is conflict with the inav config settings then this should be addressed in the design of the program.

sensei-hacker, 2024-12-01:
This isn't quite what you're asking for, but -

> In order to copy settings from one model to another

The fastest way to do that is to run "diff", then open the diff file. Open the new model and either load the whole diff or copy-paste whatever sections you want.

That's much quicker and less error-prone than clicking every setting in the GUI.

cptX, 2024-12-01:
@sensei-hacker this is off topic unfortunately. Having more than 1 instances helps visualize a lot of settings, which from diff are not understandable. Also not all settings can be setup from the command line.

cptX, 2024-12-01:
I found a solution. Run every instance from the terminal with the following line

/opt/inav/inav-configurator/inav-configurator --user-data-dir=/home/user/inav1
/opt/inav/inav-configurator/inav-configurator --user-data-dir=/home/user/inav2
/opt/inav/inav-configurator/inav-configurator --user-data-dir=/home/user/inav3

This will create of course 3 different config directories, but configuration of inav-configurator actually is not needed for anything.

I would like though this functionality to be available by default so please don't close this feature request.

MrD-RC, 2024-12-01:
> Also not all settings can be setup from the command line.

Actually, it's the opposite. Every setting is available in the CLI. Not everything is available in Configurator. If you hover over something in the Configurator. It should show you the CLI parameter.

MrD-RC, 2024-12-01:
Of course a diff can be interpreted and copied. 

https://github.com/iNavFlight/inav/blob/master/docs/Settings.md allows interpretation. Copying is simply a matter of copying and pasting the parameters you want to copy. If you want the same settings on multiple installations. Selecting parts of the diff to be pasted in to all setups is by far the quickest method.

MrD-RC, 2024-12-01:
Just perform a diff all. Find the adjustments. Then paste in to the CLI of the new flight controller, press enter, then save. 

I have done that many times for adjustments.

b14ckyy, 2025-01-18:
Since this is a very rare edge use case that would need major changes in the configurator data handling, this won't be implemented. Also I can see you found a solution for yourself. Going to close.

closed 2025-01-18 by b14ckyy