iNavFlight/inav-configurator #2278: 8.0 RC2 Hanging on "waiting for data" page

dcan999, 2024-12-07:
Speedybee F405 Mini, INAV 8.0

90% of the times I click reboot and save after adjusting something, in any tab, it will just hang on the "waiting for data" page and I have to unplug/replug the USB.

0crap, 2024-12-08:
Same here, usually clicking the Disconnect/Connect button a few times solves the issue.
Does not happen with the 7.1.2 Configurator.

[image]

dcan999, 2024-12-08:
It's been hanging up for me since 6.0 on full chip erase/resets but only when I initially do the reset/erase...on 8.0 it's hanging after almost every "save and reboot" I do regardless of how much I had changed. I also tried it today on a Matek 405WMN and it does the same.

0crap, 2024-12-10:
> Try clicking the "Flush settings cache" button in the main application settings box, Also restart Configurator after flashing and flushing the settings cache.

This could work as a temporary work-around.
As with the v7 Configurator, this should not be needed to do.

b14ckyy, 2025-01-18:
I can confirm that this issue still occasionally appears, in my case only through slow wireless connections like Bluetooth or mLRS Data Links. 
A quick disconnect and reconnect solves it but not reliably repeatable.

zahachtah, 2025-02-17:
I "solved" the problem after seeing inav 8-0 only wants 8-0 firmwares for the FC, and as I wrote in previous post: INAV 8.0, FC FIRMWARE7.1.2....

so not an issue using INAV 7.x

@b14ckyy do you want me to text the loop time fix for the INAV 8.0, FC FIRMWARE7.1.2 combo or is that then irrelevant?

pr0llux, 2025-10-27:
I have this issue with INAV 8.01 (both firmware and configurator).