iNavFlight/inav-configurator #2280: INAV8 RC2 Resets to Connection screen

MiguelitoBeard, 2024-12-08:
I am testing mLRS and INAV 8 RC2. I am using mLRS on a Matek 915 RX/TX system and the latest 1.3.03 dev firmware for such. The connection is really quite sluggish, when compared to MWP Tools that I also testing, but when I click the GPS tab in INAV more often than not INAV Configurator resets to the home connection screen.

MiguelitoBeard, 2024-12-08:
At the suggestion of Mark Hoffman I have tried this and yes RC2 Configurator. Problem persists. "I assume you also use INAV Configurator RC2 and not only firmware?
Can you please delete the content of the following folder:
C:\Users\USERNAME\AppData\Roaming\INAV Configurator
But maybe leave the config.json in there and don't delete it to not lose your settings and Bing API Key."

b14ckyy, 2025-01-18:
fixed with RC3 and later and mLRS 1.3.04 or later.

closed 2025-01-18 by b14ckyy