iNavFlight/inav-configurator #2286: 8 rc3 configuratorthere is no airframe rending.

and-sh, 2024-12-16:
After latest update on Debian 13 there is no airframe. On windows all work fine.
[image]
[image]

In alignment tab the same.

I guess this is not very urgent because Debian 13 will be released in about a year.

sensei-hacker, 2024-12-16:
That would probably be your 3D rendering / GPU drivers and configuration. If you have an Nvidia card with their driver,  you have to compile the proprietary Nvidia driver for each kernel update.

Since you're running a Debian branch that's a year before release, I probably don't need to tell you how to use journalctl to check the logs, etc. :)

and-sh, 2024-12-16:
[image]

stronnag, 2024-12-16:
This happens on modern Linux systems due to upstream issues in Electron. 
Unlikely that INAV can fix it.
It's happened ever since the configurator Electron migration.

and-sh, 2024-12-17:
Thank you. Hope they fix that soon.

Scavanger, 2024-12-18:
Fixed in #2285

and-sh, 2024-12-18:
Can I test this?
[image]

Scavanger, 2024-12-18:
> Can I test this? 

Sure, go to the PR -> Checks -> Artifacts and download the installer/archive

and-sh, 2024-12-18:
Yep, work fine. 
[image]

But there are some warnings in console

[198545:1218/193441.898004:ERROR:gl_surface_presentation_helper.cc(260)] GetVSyncParametersIfAvailable() failed for 1 times!
[198545:1218/193453.065980:ERROR:gl_surface_presentation_helper.cc(260)] GetVSyncParametersIfAvailable() failed for 2 times!
We're closing...
(node:198509) UnhandledPromiseRejectionWarning: TypeError: js.stopAll is not a function
    at App.<anonymous> (file:///usr/lib/inav-configurator/resources/app/.vite/build/main.js:11684:42)
    at process.processTicksAndRejections (node:internal/process/task_queues:95:5)
    at emitUnhandledRejectionWarning (node:internal/process/promises:298:15)
    at warnWithErrorCodeUnhandledRejectionsMode (node:internal/process/promises:406:5)
    at processPromiseRejections (node:internal/process/promises:470:17)
    at process.processTicksAndRejections (node:internal/process/task_queues:96:32)
(node:198509) TypeError: js.stopAll is not a function
    at App.<anonymous> (file:///usr/lib/inav-configurator/resources/app/.vite/build/main.js:11684:42)
    at process.processTicksAndRejections (node:internal/process/task_queues:95:5)

Scavanger, 2024-12-18:
> But there are some warnings in console

Still a Bug in electron (chromeium): https://github.com/electron/electron/issues/32760

Seems to work on some systems, on some not.
We have to disable GPU acceleration on Linux to get it work on all systems, but makes the 3D rendings sluggish. :(

and-sh, 2024-12-18:
Not at all. 
But the propellers are not positioned correctly. All counterclockwise.
In Betaflight too.

b14ckyy, 2025-01-18:
Since main issue seems to be fixed and further improvements are on the Linux side, will close here.

closed 2025-01-18 by b14ckyy