iNavFlight/inav-configurator #2288: Demo Mode does not work in all versions of 8.0.0-RC1, RC2, and RC3

ToshihiroMakuuchi, 2024-12-20:
Demo Mode does not work in all versions of 8.0.0-RC1, RC2, and RC3.

I installed the Windows version (INAV-Configurator_win32_x64_8.0.0.msi), but the Demo Mode does not work.
Does it function properly in your environment? Please share your experience.
When I press "Connect," it seems to freeze and does not proceed further.

Unfortunately, I do not have the environment to test the Linux or macOS versions.
I have confirmed that Demo Mode works correctly in INAV-Configurator_win64_7.1.2.exe.

mmosca, 2024-12-20:
Have you tried the latest configurator nightly build? A potential fix for that was merged recently.

ToshihiroMakuuchi, 2024-12-22:
Thank you for providing the information.
I haven't tried the latest nightly build yet.
I'll test it now to check its functionality.

closed 2024-12-22 by ToshihiroMakuuchi