iNavFlight/inav-configurator #2293: Please add ARM64 linux build

Botspot, 2024-12-30:
M1 Macs running Alpine Linux, ARM Chromebooks running the debian linux container, and Single Board Computers could all run this software fairly easily. I just tried building it on my Raspberry Pi 5 using the build instructions on the README, and it runs great.

Would you consider adding an ARM64 linux .deb CI release at a minimum? If so, I should be able to add this to [the Pi-Apps store](https://github.com/Botspot/pi-apps), which already has Mission Planner on it.

mmosca, 2024-12-31:
The plan is to add it as soon as GitHub adds arm64 Linux runners. Will probably be available for 8.1, given the current GitHub roadmap.

b14ckyy, 2025-01-18:
Added recently. Available in Nightly build here https://github.com/iNavFlight/inav-configurator-nightly/releases/tag/v20250118.101

Basic function tests including SITL and Demo Mode successful on Raspberry Pi with Raspberry-OS Bookworm.

closed 2025-01-18 by b14ckyy