iNavFlight/inav-configurator #2294: 8.0.0 RC3 - Logic conditions Screen Disconnect when saving

Itay-Sharoni, 2024-12-31:
Hi, I tested version 8.0.0 RC3

If i go to Mixer > Logic Conditions (at bottom) > do anything and click save > it will not save, and disconnect INAV Configurator, in the log i see: Unexpected error: MSPChainerClass is not defined

If im doing the changes from CLI using logic command it works fine.

[image]

[image]

b14ckyy, 2025-01-18:
Can you please try again with the latest Nightly build? 
https://github.com/iNavFlight/inav-configurator-nightly/releases/tag/v20250118.101

mmosca, 2025-01-21:
I managed to get this issue, but only when launching it from the mixer tab, not from the programming tab.

@b14ckyy did you get it work from the mixer tab?

b14ckyy, 2025-01-21:
Thanks for the reminder. Yes I was able to replicate last night. When saving the logic programming overlay in mixer tab it crashes. From Logic Programming tab directly its fine. 
I guess it has something to do with the overlay itself?