iNavFlight/inav-configurator #2334: Unable to get proper motor signal on output S11

ViorelX, 2025-02-01:
I am setting up INav 8 on **FC Speedybee f405 Wing Mini**. Up to now, I have used signal output 1 for motor and S3-S8 for servos.
Now I want to use S1-S6 for servos (S1, S6 flaps, S2-S5 ailerons, elevator, rudder), because the **servo autotrim** function works only for the 2-5 servos. Thus the motor should be put elsewhere and I chose S11.
All S1-12 outputs of this FC are PWM and DShot capable, as its handbook says.
I have configured servos and motor in the mixer tab as follows:

[image]

Unfortunately, I cannot start the motor. I tried all types of signals: PWM, OneShot, MultiShot, DShot. I get a signal on S11 output, but the motor can't be started (the ESC is properly set up).
After a lot of tryings I managed at one point to start the motor on S11 (with all types of signal), but as soon as I made some other settings on FC (not related to servos and motor), the problem of starting the motor on S11 came back again.

What have I do:
I put the motor on S8 and it works:

[image]

However, this it's not very convenient. In order to get motor signal on S8, all S8, S9, S10 become motor outputs (being driven by the same Timer). If you want add other servos, you cannot use S9, S10 anymore.
S11 is the most convenient motor output, since it is served by the Timer12, which serves a single output and all other outputs remain free for other uses.

sensei-hacker, 2025-02-01:
How we you trying to run the motor?
With the throttle stick in the transmitter? From the Outputs tab?

> S8, S9, S10 become motor outputs

Yeah, which leaves 9 servo channels.
Pitch
Roll
Yaw
Flaps
and five more

Do you need more than 9 servo channels?

ViorelX, 2025-02-01:
To run the motor I used throttle stick (but this requires arming) a potentiometer on some RC channel and the output tab.
Yes, 9 channels remain left.
6 are already in use:
pitch
2x roll
1 yaw
2x flaps

1 has the LED function.
There are only 2 left free.

At this moment I don't need more channels. But it doesn't make sense to waste 3 channels for a single motor and it is clear there is a bug if you cannot use S11 as a motor channel, excepting sometimes while the debug process.

sensei-hacker, 2025-02-01:
Please test using the Outputs tab, as that will eliminate a dozen different things that could block arming.

Dhsot doesn't require calibration; Multishot and Standard protocol require calibration via the Outputs tab (not the transmitter).

If we don't figure out what's wrong, and later you need more than three more servo channels, you would also have the option of not duplicating both the flaps and the yaw. That would give you five free servo channels.