iNavFlight/inav-configurator #2338: New 8.0.0 version of Configurator doesn't work on mac m1.

bobirbek1, 2025-02-03:
Today I install new INav configurator 8.0.0 and I wasn't able to run programm. I recieve below error. 

[image]

bobirbek1, 2025-02-03:
Yes I downloaded darwin_arm64

bobirbek1, 2025-02-03:
[image]

This was when I tried to run on terminal

tradom73, 2025-02-07:
> <img alt="Image" width="1080" src="https://private-user-images.githubusercontent.com/50242958/409160710-2cfa4730-eed5-41e1-aeb5-a6d3cf809d0a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mzg5NTUxMDMsIm5iZiI6MTczODk1NDgwMywicGF0aCI6Ii81MDI0Mjk1OC80MDkxNjA3MTAtMmNmYTQ3MzAtZWVkNS00MWUxLWFlYjUtYTZkM2NmODA5ZDBhLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAyMDclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMjA3VDE5MDAwM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTMzYmUyMmUzNWNmZjYzNWE3MjUzMmE3ZTBlYzcwMTBlYjliNjA2NzkyN2UyMDRjZTFjMGRjMmY0ODA3NGEwOGImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.ucNa9-9MMmGZO0lsDxnlPhhCn6ntx6WMmV3F52Em16s">
> This was when I tried to run on terminal

I have same problem, did you solve it?

Eka157, 2025-02-07:
im running into the same thing...

vanderkl, 2025-02-08:
I am having the same problem and get the same results trying xattr command.

mmosca, 2025-02-09:
Try deleting the files in the application folder, downloading inav 8 again and running the command before you try to run it.

Works ok for me.

[image]

What MacOS version are you using? I think the binaries were built on Sequoia (15.x).

Snyperx, 2025-02-10:
I am on 15.3 and getting the message after a fresh download and install.

halfmanbear, 2025-02-11:
I fixed it with the following;

```
brew install node

cd /Applications/INAV\ Configurator.app/Contents/Resources/app

npm install

```

tradom73, 2025-02-11:
> brew install node
> 
> cd /Applications/INAV\ Configurator.app/Contents/Resources/app
> 
> npm install

didn't help ;/
od 15.3

mmosca, 2025-02-14:
Did it run, despite the error?
At leas one user on discord, with the same issue said it launches fine, even with the complaints.

sensei-hacker, 2025-02-14:
> <img alt="Image" width="1080" [...]

One person reported that they got that error message - then ran the appl

tradom73, 2025-02-19:
I can't run it, anyway I try to do it it gives the error and doesn't open the app

Mauronic, 2025-03-10:
I get the same damaged error message. Using 8.0.0 Darwin Arm 64 with a M4 Pro running OSX 15.3.1.

I tired this and it didn't work

> brew install node
> 
> cd /Applications/INAV\ Configurator.app/Contents/Resources/app
> 
> npm install

ph0nebooth, 2025-03-26:
Running this in Terminal will resolve the issue.

xattr -cr "/path/to/your/INAV Configurator.app"

Pavelkoen, 2025-06-24:
Nothing is working, competelly useless OS for INAV

Tanmax-Payne, 2025-11-13:
xattr -d com.apple.quarantine "/Applications/INAV Configurator.app"
this solved it