iNavFlight/inav-configurator #2340: Version 8.0.0 not working on Ubuntu

MichaelThamm, 2025-02-05:
I tried out inav-configurator .deb (8.0.0) package with a SpeedyBee F405 V3 flight controller. I tried to flash the 8.0.0 (stable) SPEEDYBEEF405V3 firmware onto my FC. I get this log output right after pressing the Flash Firmware button (connected on /dev/tty/ACM0).

```
2025-02-04 @ 20:55:57 -- Failed to open USB device!
2025-02-04 @ 20:55:57 -- Are udev rules installed correctly? See docs for instructions
```
It seems that the flash never completes cleanly because after waiting some time and rebooting the FC, I can see the /dev/tty/ACM0 interface become available in the list but I still cannot connect to the FC.

I tried to follow the troubleshooting:

```
If you have lost communication with your board follow these steps to restore communication:
- Power off
- Enable 'No reboot sequence', enable 'Full chip erase'.
- Jumper the BOOT pins or hold BOOT button.
- Power on (activity LED will NOT flash if done correctly).
- Install all STM32 drivers and Zadig if required (see [USB Flashing](https://github.com/iNavFlight/inav/blob/master/docs/USB%20Flashing.md) section of INAV manual).
- Close configurator, Close all running chrome instances, Close all Chrome apps, Restart Configurator.
- Release BOOT button if your FC has one.
- Flash with correct firmware (using manual baud rate if specified in your FC's manual).
- Power off.
```

but this just resets to the default firmware and when I try to connect in that state I get [this scenario](https://github.com/iNavFlight/inav/issues/4093) which is expected.

Where am I going wrong or is this a bug?

sensei-hacker, 2025-02-05:
Are udev rules installed correctly? See docs for instructions

MichaelThamm, 2025-02-05:
https://github.com/iNavFlight/inav/blob/f00cd9ee37fa54f8674b17117bd88e34bf304e0b/docs/USB%20Flashing.md#platform-specific-linux

Yeah I just saw this based on my log output. Thanks for the quick answer. This works and will close the issue!

closed 2025-02-05 by MichaelThamm