iNavFlight/inav-configurator #2342: INAV Configurator 8 version for Windows

mickey2011, 2025-02-09:
What is the current file name to download INAV Configurator 8. I have Windows 11 on a x64 computer.

The instructions are telling me I need the "win32_x64 version, but I cannot find a file name that matches the instructions.

Thanks

sensei-hacker, 2025-02-09:
Grat question! You may need to click at the end of the list where it says "show more".
GitHub automatically puts them in alphabetical order, and as it happens Windows / Win32 is at the end in alphabetical order.

mickey2011, 2025-02-09:
> Great question! You may need to click at the end of the list where it says

Thanks

Peter H

mmosca, 2025-02-09:
I know it sounds like you found the download already, but in hopes people find this answer in search:

[image]

You just need to click on Show all 16 assets

[image]

closed 2025-02-09 by mmosca