iNavFlight/inav-configurator #2355: It is not possible to change the position elements

JerryLutor, 2025-03-08:
It is not possible to change the position
Crosshairs
Artifical horizont
Horizond sidebars

These elements do not have an indicator on the screen for which they can be moved on the screen.

In this mode they are set not in the center of the screen in the glasses.

[image]

Inav 8.0.0
Inav Configurator 8.0.0

board - speedyby wings 405
VTX - DJI O3
osd mode - DJIWTF

mmosca, 2025-03-08:
This is by design. 

They are centered in the canvas. 

If they are off center you may be using the wrong osd type. 

Dji_wtf is for goggles v2 only and has a bigger canvas than the o3 normally uses. 

For googles 2 and newer, with latest firmware, you should use avatar or dji native.

closed 2025-03-08 by mmosca

JerryLutor, 2025-03-08:
There is no such parameter as “dji native” in the configurator. What should I specify there? What do I need to specify there? 
@mmosca