iNavFlight/inav-configurator #2356: CLI parameters and units

vprheli, 2025-03-16:
What about add units for parameters in CLI?

sensei-hacker, 2025-03-16:
What's the use case?

vprheli, 2025-03-16:
It is good to know if the parametr is for instance 1000 is in miliseconds,
microseconds or meter, centimetr...

It is not stated anywhere that, for example, all lengths are in cm,
time in milliseconds (which is not the case), etc.

sensei-hacker, 2025-03-16:
Okay, I understand. I had misunderstood what you meant before.

MrD-RC, 2025-03-16:
That's what this document is for https://github.com/iNavFlight/inav/blob/master/docs/Settings.md

Adding that information to CLI would just take up a load of flash space.

sensei-hacker, 2025-03-16:
If you want to try adding it to the CLI autocomplete in Configurator, you can look at adjusting this file:

https://github.com/iNavFlight/inav-configurator/blob/master/js/CliAutoComplete.js