iNavFlight/inav-configurator #2357: Mission Control in Configurator not updating SafeHome on screen data

NorthIceMan, 2025-03-24:
## Current Behavior

## Steps to Reproduce
,  and go to Mission Control.  You should see a SafeHome on the map (if not Load a Mission File from Action Menu).  Note the "Edit Safehome x" on left panel shows the SafeHome number in place of x.
Now (from Action Menu) .  Note the SafeHome icon on map has changed to the relevaLoad a different mission filent SafeHome.  However the Edit Safehome panel to left still has the previously loaded mission file's SafeHome. 

1. Run iNAV Config
2. connect to FC
3. go to Mission Control
4. Load a Mission File from Action Menu
5.  Note the "Edit Safehome x" on left panel shows the SafeHome number in place of x (which matches the SafeHome icon on map). Note the SafeHome number
6.  Load a different mission file with a different SafeHome (different location)
7.  Note the new SafeHome icon and its number on the map
8.  Note however that the Edit Safehome panel to left still has the previously loaded mission file's SafeHome - not the one shown on the map. 

## Expected behavior
When a different Mission file is loaded (having a different SafeHome) the Edit Safehome panel at left should display the Safehome info for the icon on the map not the previous Mission's SafeHome's info.

## Suggested solution(s)
1.  Update the Edit Safehome data to match the Safehome icon on map
2. OR  Notify user that a new SafeHome may need be loaded by clicking Load eeprom SafeHome icon in the Safe Home Manager title line at left.

## Additional context
Note: Even though I sometimes fly different Missions using the same SafeHome I do have a separate SafeHome for each different Mission Location I fly. When dealing with multiple SafeHome's this can cause mis-understandings with AutoLand.

---
# version
# INAV/SPEEDYBEEF405WING 8.0.0 Jan 21 2025 / 16:16:30 (ec2106af) 
# GCC-13.2.1 20231009
- FC Board name and vendor: SPEEDYBEEF405WING
- INAV version string: 8.0.0 Jan 21 2025 / 16:16:30 (ec2106af)

MrD-RC, 2025-03-24:
Safehomes are independent to missions. No matter which mission is loaded. The safehomes should be the same. 

This is working as expected.

closed 2025-03-24 by MrD-RC

NorthIceMan, 2025-03-24:
Hey Darren - thanks for the attention to this.
Sorry to see you closed this but no worries, I will respond to the 4 other
chaps who had the same issue.

I have missions which are up to 10 miles apart.  My SafeHomes are not
"always the same".  I say this because you responded "Safehomes are
independent to missions. No matter which mission is loaded. The safehomes
should be the same."
We realize that SafeHomes are independent to missions.  The problem is - on
a failsafe which of the SafeHomes shown in iNAV is used - the one the iCON
displays or the Coordinates displayed in the left sidebar editor? The User
Interface is displaying an icon for one location but the sidebar is
displaying another in numbers that is many miles away and its not evident
to the user because one is in numerals while the other is just an icon.
I do understand though that if you always fly from same location that is
not as much an issue.

Ron

MrD-RC, 2025-03-24:
It was closed because there is no error. 

If within range, the safehome closest to the arming location is used.

There is a maximum distance from the arming location. If the all safehomes are outside of this distance. They will not be used at all. Safehomes and INAV's fixed wing landing are not intended to be used at remote locations. They should be within walking distance of the arming point. This is all documented. The default maximum distance for a safehome from the arming point is 200m. The maximum is 650m.

https://github.com/iNavFlight/inav/blob/master/docs/Safehomes.md

NorthIceMan, 2025-03-24:
Yep--all good info for anyone who doesn't know these things.
Now, why was it showing coordinates for a SafeHome miles away in the SideBar when I loaded the Mission?  Thats the question-I have no other question.
All my parameters in Advanced Tuning are within the guidelines , why would it keep coordinates for a SafeHome miles from the previously loaded Mission when it shows the correct icon on the map.  Am I not being clear enough?

MrD-RC, 2025-03-24:
Safehomes have nothing to do with the loaded mission. Mission Control will load and show all safe homes. I could plan a safehome the other side of the world and it will still show up.