iNavFlight/inav-configurator #2359: Default timer output setting does not allow to activate motor resource OMNIBUSF4V6

kreidler, 2025-03-26:
Upgrading from INAV 4.1 to 8.0.0 (tested same behaviour on 7.1.2) on OBV6.

Flashing 8.0.0 or 7.1.2
With or without selecting default values
Mixer tab TIMER 8 is set to MOTOR
Mixer tab TIMER 2/3/4 are set to AUTO

Applying a QUAD X by defaults or mixer preset will show 3 motors on output mapping only.
Additionally as long as TIMER 8 is not set to AUTO or in a combination like MOTOR on TIMER2/4/8 or TIMER3/8 no motor pins are assigned on resources and always 3 motors will be shown on output mapping.

Expected behaviour:
When applying a mixer preset or default values (7" mini quad etc.) motor pins should be enabled without further action.

mmosca, 2025-03-26:
did you save and reboot, after changing the mixer?

kreidler, 2025-03-26:
Yes, I assume about 50 times at least to circle in the issue. BTW: On the mixer wizard a save and reboot is included / cannot  be avoided.

sensei-hacker, 2025-03-26:
Yep, this is the result with erase settings and applying the default quad.

[image]

I'll see what I can figure out on it.

sensei-hacker, 2025-03-26:
The workaround for now is to set all three timers to motors, save and reboot.
Of course we would like to find a better long term fix.

sensei-hacker, 2025-03-27:
I've investigated this issue some and expect to make a PR soon to address it.

kreidler, 2025-03-27:
Thanks for investigating.
My knowledge to the timer settings is very limited. So I switched all timers to AUTO which works for me. Can I leave it like it is or is there any requirement for setting the timers to MOTOR in this case. No servos or LED connected to the quad yet.