iNavFlight/inav-configurator #2366: inav 6.0

davifelix, 2025-04-19:
Hello everyone.......I know it's an unusual request, but I absolutely need to flash inav 6.0 I had a perfectly calibrated and working vehicle......I saved the dump and installed inav 8.0 now it doesn't fly well anymore and I want to go back to inav 6 to have the vehicle working perfectly again. The problem is that the configurator doesn't let me go back to the version I need. Is there a way to solve it??

daijoubu, 2025-04-19:
Go to the Inav repository and download the file from the 6.x release.  You'll need a 6.x configurator as well.
Here's the last release in the 6.x series.

davifelix, 2025-04-20:
Quindi è ancora possibile..... Non ho ben capito come fare, ma farò qualche tentativo...... Grazie

sensei-hacker, 2025-05-26:
Click here
https://github.com/iNavFlight/inav/releases?page=2#:~:text=6.1.1

Scorll down half a page or so and click "Assets".

closed 2025-05-26 by sensei-hacker