iNavFlight/inav-configurator #2369: Geozone causing issues with missions

breadoven, 2025-04-20:
Trying to add multimission in Mission Control. Works OK for a while then the following errors start appearing which causes the whole thing to crash.

2025-04-20 @ 11:07:44 -- Unexpected error: Cannot read properties of null (reading 'geozoneCount')

2025-04-20 @ 11:07:52 -- Unexpected error: Cannot read properties of null (reading 'geozoneCount')

2025-04-20 @ 11:07:54 -- Unexpected error: Cannot read properties of null (reading 'geozoneCount')

2025-04-20 @ 11:08:00 -- Unexpected error: Cannot read properties of undefined (reading 'getLonMap')

2025-04-20 @ 11:08:03 -- Unexpected error: Cannot read properties of null (reading 'geozoneCount')

2025-04-20 @ 11:08:06 -- Unexpected error: Cannot read properties of undefined (reading 'getLonMap')

2025-04-20 @ 11:08:11 -- Unexpected error: Cannot read properties of null (reading 'geozoneCount')

2025-04-20 @ 11:08:19 -- Unexpected error: Cannot read properties of undefined (reading 'getLonMap')

2025-04-20 @ 11:08:25 -- Unexpected error: Cannot read properties of null (reading 'geozoneCount')

2025-04-20 @ 11:08:30 -- Unexpected error: Cannot read properties of undefined (reading 'getLonMap')

2025-04-20 @ 11:08:34 -- Unexpected error: Cannot read properties of undefined (reading 'getLonMap')

2025-04-20 @ 11:08:38 -- Unexpected error: Cannot read properties of undefined (reading 'getLonMap')

2025-04-20 @ 11:08:40 -- Unexpected error: Cannot read properties of undefined (reading 'getLonMap')

2025-04-20 @ 11:08:45 -- Unexpected error: Cannot read properties of undefined (reading 'getLonMap')

2025-04-20 @ 11:09:16 -- Unexpected error: Cannot read properties of undefined (reading 'getMultiMissionIdx')

2025-04-20 @ 11:09:17 -- Unexpected error: Cannot read properties of undefined (reading 'getMultiMissionIdx')

It's also noticeable that adding a mission is erratic. Several mouse clicks are required to do things such as adding WPs, selecting WPs etc. All very sluggish and erratic.

I'm guessing this is related to Geozones.

closed 2025-05-05 by breadoven

wumfi, 2025-05-14:
As per mention from @stronnag, I also have this issue.  I added more detail there.  Would you like me to paste that content here if it would be helpful?

wumfi, 2025-05-14:
Video showing the issues I'm having: [MEGA link](https://mega.nz/file/E0BzxRJa#wQhyn75I9s3uq5ttsn_8OghLwnULDYyXTEmSZIpSZSg)

stronnag, 2025-05-14:
The issue is already fixed in https://github.com/iNavFlight/inav-configurator/pull/2371 as mentioned above.