iNavFlight/inav-configurator #2376: Do you plan to support Flatpak or AppImage?

652436962, 2025-05-20:
Do you plan to support Flatpak or AppImage?

sensei-hacker, 2025-05-20:
I don't see any benefit to Flatpak or AppImage over the four existing formats available for Linux. Any reason you can't use the zip? Or the room or deb package?

Phrozenn1, 2025-07-17:
The benefit of flatpak is that it's the closest thing to a universal format for linux as you get. It supports all distros and it would be a godsend for the users that might struggle installing stuff manually.

sensei-hacker, 2025-07-17:
> The benefit of flatpak is that it's the closest thing to a universal format for linux as you get. It supports all distros and it would be a godsend for the users that might struggle installing stuff manually.

Which distro are you having trouble on?

krabodyan, 2025-08-02:
I am also interested in Nix support. Since nixpkgs only has version 5.1. Perhaps inav could have its own Nix Flake