iNavFlight/inav-configurator #2398: Support for QMC5883P

synersignart, 2025-08-02:
Recently we been getting versions circulating over the market of the QMC5883P said to be the replacement of the QMC5883
and seeing Ardupilot already implemented this months ago. seeing that new FC or GPS would come package with these in months to come

here is the comparison of the 2 

**QMC5883**
```
#if defined(QMC5883)

#define MAG_ADDRESS 0x0D
#define MAG_DATA_REGISTER 0x00
//REG CONTROL

#define MAG_CTRL_REG1 0x09
#define QMC_MODE 0xC1 // 11 00 00 01OSR=64(11); Range=2G(00); ODR=10Hz(00); MODE=Cont(01)
#define MAG_CTRL_REG2 0x0A

void Mag_init() {
  i2c_writeReg(MAG_ADDRESS,0x0B,0x01);
  i2c_writeReg(MAG_ADDRESS,MAG_CTRL_REG1,QMC_MODE);
}
```

**QMC5883P**

```
#if defined(QMC5883P)

#define MAG_ADDRESS 0x2C
#define MAG_DATA_REGISTER 0x01
//REG CONTROL

#define MAG_CTRL_REG1 0x0A
#define QMC_MODE 0x09 // 11 00 00 01OSR=64(11); Range=2G(00); ODR=10Hz(00); MODE=Cont(01)
#define MAG_CTRL_REG2 0x0B

void Mag_init() {
  i2c_writeReg(MAG_ADDRESS,0x0D,0x40);
  i2c_writeReg(MAG_ADDRESS,0x29,0x06);
  i2c_writeReg(MAG_ADDRESS,MAG_CTRL_REG1, 0xCF);
  i2c_writeReg(MAG_ADDRESS,MAG_CTRL_REG2, 0x00);   
  i2c_writeReg(MAG_ADDRESS,MAG_CTRL_REG1,QMC_MODE);
}
```
Data sheet

[https://www.qstcorp.com/upload/pdf/202202/%EF%BC%88%E5%B7%B2%E4%BC%A0%EF%BC%8913-52-19%20QMC5883P%20Datasheet%20Rev.C(1).pdf](url)

Recent reports from Arduino forum

[https://forum.arduino.cc/t/are-all-new-gy-270-board-magnetometers-now-fake/1379276](url)

sensei-hacker, 2025-08-02:
Do you happen to have one, for testing?

synersignart, 2025-08-02:
> Do you happen to have one, for testing?

oh yes and I have run it on a multiwii , I yet to check for INAV as I yet to understand how INAVs drivers read the data to port correctly