iNavFlight/inav-configurator #2402: This firmware version is not supported. This version of Configurator supports firmware from 7.0.0 to 8.0.0 (excluded)

mbarcus23-oss, 2025-08-20:
I got locked out of flashing a Goku 745 Nano maybe because i didnt have full erase on.
Finally got it flashing and install Cli but i cant restart configurator
Ive got latest Firmware and latest Config 7 as on;y use windows 7 cant go to 8.
Initially got in using bootloader but after using STM32 to flash i got into Comfig
without bootloading
Also used Zadig as i thought was getting USB issues
Hope theres a fix

sensei-hacker, 2025-08-20:
The message in your subject line means that the version of Configurator doesn't match the version of INAV.

If you want to tmuse INAV 8, use Configurator 8.
If you want to use INAV 7, use Configurator 7.

sensei-hacker, 2025-08-20:
> config 7 with updated Hex

If you did Configurator 7 with INAV 8, we would expect you would get that message.