iNavFlight/inav-configurator #2406: Flashing Issue

edwardfurn-svg, 2025-09-20:
Im having real issues flashing the software onto a speedy bee f405 wing controller, I am able to do the erasing process but the flashing gets stuck. I'm currently using lynex cause I have a chrome book. Chat GPD keeps saying the file is to big 1.7 mb, but that's what keeps getting downloaded. if some one could give me a hand that would be great. Thanks in advance Eddie.

sensei-hacker, 2025-09-20:
What does the error message say?

Also, you might try a fresh USB data cable.
Inexpensive cables often get flaky after some use, because they are made with the tiniest of wires.

edwardfurn-svg, 2025-09-20:
when the boot button is pressed it picks it up when its not nothing seems to happen.  Thanks for your help man.

edwardfurn-svg, 2025-09-20:
This is now the message 2025-09-20 @ 18:02:06 -- Running - OS: Linux, Chrome: 120.0.6099.216, Configurator: 8.0.1

2025-09-20 @ 18:02:07 -- Loaded release information from GitHub.

2025-09-20 @ 18:02:12 -- Attempting automatic target selection

2025-09-20 @ 18:02:12 -- Cannot prefetch target: Flight Controller in DFU

2025-09-20 @ 18:02:30 -- USB device successfully opened

but its just getting stuck on flashing

edwardfurn-svg, 2025-09-21:
Hi thanks for your help mate, chat GPD sorted it out, I managed to flash it without i nav and it now seems to be working correctly thanks for your help. I have a feeling it was having issues with my slow chrome book. Cheers for your help anyway.
Eddie