iNavFlight/inav-configurator #2407: Aircraft animation does not work on Linux

ivlis, 2025-09-27:
It used to work before, but now the aircraft animation is just empty. The same is true for the alignment tool

[image]

[image]

I tried aur packaged version and the version from releases - the same bug.

sensei-hacker, 2025-09-27:
Those animations use the 3D hardware on your graphics card. Did you recently update your GPU driver?

Or trying to use the proprietary Nvidia driver, so you have cuda?

ivlis, 2025-09-27:
I have an AMD GPU
```
03:00.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Navi 32 [Radeon RX 7700 XT / 7800 XT] (rev c8)
```

```
pacman -Qs radeon
local/lib32-vulkan-radeon 1:25.2.2-2
    Open-source Vulkan driver for AMD GPUs - 32-bit
local/linux-firmware-amdgpu 20250808-1
    Firmware files for Linux - Firmware for AMD Radeon GPUs
local/linux-firmware-radeon 20250808-1
    Firmware files for Linux - Firmware for ATI Radeon GPUs
local/vulkan-radeon 1:25.2.2-2
    Open-source Vulkan driver for AMD GPUs
```

sensei-hacker, 2025-09-27:
It looks like that was compiled by distro about six weeks ago.
Do you know when it was updated on your system?

ivlis, 2025-09-27:
```
pacman -Qi local/vulkan-radeon
Name            : vulkan-radeon
Version         : 1:25.2.2-2
Description     : Open-source Vulkan driver for AMD GPUs
Architecture    : x86_64
URL             : https://www.mesa3d.org/
Licenses        : MIT AND BSD-3-Clause AND SGI-B-2.0
Groups          : None
Provides        : vulkan-driver
Depends On      : expat  gcc-libs  glibc  libdrm  libelf  libx11  libxcb  libxshmfence  llvm-libs  spirv-tools  systemd-libs  vulkan-icd-loader  wayland  xcb-util-keysyms
                  zlib  zstd
Optional Deps   : vulkan-mesa-layers: additional vulkan layers
Required By     : lib32-vulkan-radeon  steam  zed
Optional For    : sdl3  vulkan-icd-loader
Conflicts With  : None
Replaces        : None
Installed Size  : 14.31 MiB
Packager        : Jan Alexander Steffens (heftig) <heftig@archlinux.org>
Build Date      : Sun 07 Sep 2025 05:24:29 PM EDT
Install Date    : Sat 13 Sep 2025 09:35:40 PM EDT
Install Reason  : Explicitly installed
Install Script  : No
Validated By    : Signature
```

Last time i saw the aircraft working was before 09/13.

sensei-hacker, 2025-09-27:
So if I understand you correctly, you haven't seen it working correctly after the driver was updated on 9/13.

But it works if you click the Mixer tab first, then go back.

It might be interesting to check journalctl to see if there are any messages that appear relevant.

b14ckyy, 2026-09-22:
Potentially fixed, no OP response. Closing

closed 2026-09-22 by b14ckyy